zephyrproject-rtos/sdk-ng #838: Clang Picolibc FP test failures on RV32F

stephanosio, 2024-11-24:
Some Picolibc floating-point tests fail on RV32 architectures with hardware FPU (F extension)

For instance, on `rv32imafc_zicsr_zifencei`:

```
122/196 picolibc / rounding-mode                           FAIL             1.42s   exit status 4
127/196 picolibc / math_errhandling                        FAIL             1.53s   (exit status 241 or signal 113 SIGinvalid)
194/196 picolibc / test-fma                                FAIL             1.63s   exit status 1
```

```
127/196 picolibc / math_errhandling                        FAIL             1.53s   (exit status 241 or signal 113 SIGinvalid)
...
Double tests:
Exceptions claim to be supported, but 0/0 doesn't raise FE_INVALID
Exceptions claim to be supported, but 1/0 doesn't raise FE_DIVBYZERO
Exceptions claim to be supported, but 5e-324*5e-324 doesn't raise FE_UNDERFLOW
Exceptions claim to be supported, but 1.7e+308*1.7e+308 doesn't raise FE_OVERFLOW
```

```
122/196 picolibc / rounding-mode                           FAIL             1.42s   exit status 4
...
testing up/down
ERROR up/down: 0 is not > 0
testing up/toward
ERROR up/toward: 0 is not > 0
testing -up/-down
ERROR -up/-down: 0 is not > 0
testing -toward/-down
ERROR -toward/-down: 0 is not > 0
```

```
194/196 picolibc / test-fma                                FAIL             1.63s   exit status 1
...
float
double
27: round FE_UPWARD -0x1.0000000000001p-1 * 0x0.0000000000001p-1022 + 0x0p+0 -> got -0x0.0000000000001p-1022 want 0x0p+0
34: round FE_UPWARD 0x0.0000000000001p-1022 * 0x0.0000000000001p-1022 + 0x0p+0 -> got 0x0p+0 want 0x0.0000000000001p-1022
35: round FE_UPWARD 0x0.8p-1022 * 0x0.0000000000001p-1022 + 0x0p+0 -> got 0x0p+0 want 0x0.0000000000001p-1022
36: round FE_UPWARD 0x1.0000000000001p-1022 * 0x0.0000000000001p-1022 + 0x0p+0 -> got 0x0p+0 want 0x0.0000000000001p-1022
37: round FE_UPWARD 0x1.0000000000001p-1021 * 0x0.0000000000001p-1022 + 0x0p+0 -> got 0x0p+0 want 0x0.0000000000001p-1022
...
```

Full test log: https://gist.githubusercontent.com/stephanosio/1934c4156f0e231651690e118ccf6207/raw/4b4463912991676e1d7e9ae2d6c71405b32a3b89/testlog.txt

keith-packard, 2024-11-24:
Welcome to clang's buggy FP support?

keith-packard, 2024-11-24:
I assume you're using llvm's compiler-rt runtime library? That's where the bugs lie -- its soft FP code doesn't support exceptions or rounding modes, unlike libgcc.

stephanosio, 2024-11-24:
> I assume you're using llvm's compiler-rt runtime library? 

Yes, from LLVM 19.1.1.

> its soft FP code doesn't support exceptions or rounding modes, unlike libgcc.

That is unfortunate. I will mark this as a known issue and disable the affected tests on the relevant targets for now.

keith-packard, 2024-11-24:
I'm downloading the new toolchain and will disable the tests once I've gotten that working. Looks like there are some minor issues with the llvm riscv toolchain which is making that a bit harder than anticipated.