zephyrproject-rtos/sdk-ng #855: ccache not working on OS X/macOS

koffes, 2025-01-19:
I discovered that ccache is not working with Zephyr SDK 0.17.0 on macOS.
Let me know if I should file this ticket elsewhere.

From the log below, ccache is installed, and is found, but there are no hits:

```
Cacheable calls:               0 / 1258 ( 0.00%)
  Hits:                        0
    Direct:                    0
    Preprocessed:              0
  Misses:                      0
Uncacheable calls:          1258 / 1258 (100.0%)
  Bad compiler arguments:    756 / 1258 (60.10%)
  Called for linking:        256 / 1258 (20.35%)
  Called for preprocessing:  124 / 1258 ( 9.86%)
  No input file:             122 / 1258 ( 9.70%)
Successful lookups:
  Direct:                      0
  Preprocessed:                0
Local storage:
  Cache size (GiB):          0.0 /  5.0 ( 0.00%)
  Files:                       0
  Hits:                        0
  Misses:                      0
  Reads:                       0
```
Another colleague of mine, noticed the same thing on a different macOS setup. He claimed that this may have happened with the transition to sysbuild.

Tested on:
Sequoia 15.0.1
SDK 0.17.0
Zephyr bleeding edge today (d3a0e16)
samples/basic/blinky can be used to test

When I get more time, I will bisect and find when this broke. I will also try different SDK versions.

kartben, 2025-01-19:
Can you try:

```
export CCACHE_IGNOREOPTIONS="-specs=* --specs=*"
```

I think we basically need to set this as a default somewhere
cc @nashif @tejlmand

koffes, 2025-01-20:
Ah! Thank you @kartben ! That worked like a charm. Roughly 3X speedup.
As you said, this would be nice to set to default if possible. Will keep issue open.

koffes, 2025-02-04:
I tried this on Windows as well, and setting the same env. var was required there as well.

keith-packard, 2025-02-04:
Can anyone explain why this is needed? Seems like the --specs option should be the same in all compiles?

kartben, 2025-02-05:
> Can anyone explain why this is needed? Seems like the --specs option should be the same in all compiles?

@keith-packard 
https://github.com/zephyrproject-rtos/zephyr/blob/main/.github%2Fworkflows%2Fcodecov.yaml#L36

Does that help?
If there is a way to fix this (maybe by providing full path to the specs file? https://stackoverflow.com/a/77878905/1051634) it would be nice :)

keith-packard, 2025-02-05:
For the SDK, we could provide the full path to the specs file. Annoying that ccache doesn't search the library path specified by the compiler. I just tried 'arm-none-eabi-gcc -print-file-name=picolibc.specs' and that happily provided me the full path name.

Damian-Nordic, 2025-02-14:
I believe the `ccache` behavior to fallback to a real compiler in such a case was added in `ccache 4.9`: https://github.com/ccache/ccache/commit/270433590edcb89b58255b4d8f28ce4b3c2ea7b1#diff-e0b15464868bd66a9c95b324c6df839ffd52af9eff5757f458326ac6602fda65R1763.