zephyrproject-rtos/sdk-ng #858: Is openocd in Zephyr toolchain too old?

binEncoder, 2025-02-06:
I have the following questions, please help me out if you are a professional who is good at debugging.
1. Is openocd( https://github.com/zephyrproject-rtos/openocd?tab=readme-ov-file ) used by zephyr no longer maintained? Its version has been stuck at 0.11.0, which is several years old.
2. If there is a debugging requirement, how can I install a higher version of openocd myself and use it in zephyr instead of using the lower version of zephyr provided.

The following are some of the processes I use to debug with pico 2.
I use a RasberryPi Pico2 and want to debug an elf built with zephyr using CMSIS-DAP( https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html ).
Below is my environment information.
OS: Ubuntu 22.04 jammy
Zephyr SDK( Toolchain ): 0.17.0
Zephyr: 4.0.99

The openocd version in the zephyr toolchain is **0.11.0**, which does not contain rp2350.cfg.
I tried to get some openocd configuration information from the raspberrypi official website( https://github.com/raspberrypi/openocd ), and after using it with zephyr, I got the following errors.
1. Can't find target/rp2350.cfg
    => I tried downloading it from the Raspberry Pi website and placing it in the appropriate location.
2. rp2350.cfg:38: Error: Unknown param: -adiv6, try one of: -chain-position, -ignore-syspwrupack, -dp-id, or -instance-id
    => I deleted the error section in rp2350.cfg.
3. rp2350.cfg:43: Error: Invalid AP number!
...

I also use the Raspberry Pi Pico SDK
( https://www.raspberrypi.com/documentation/pico-sdk/index_doxygen.html#raspberry-pi-pico-sdk ) and it can connect to my pico 2 normally.
From the pico sdk, I found that the openocd version it uses is **0.12.0**.

- Pico SDK( 2.1.0 ) + Openocd( 0.12.0 ) + Debug Probe => Connect Success
- Zephyr( 4.0.99 ) + Zephyr SDK( 0.17.0 ) + Openocd( 0.11.0 ) + Debug Probe => Connect Fail

samblenny, 2025-02-12:
I've had good results by using the Raspberry Pi version of openocd from https://github.com/raspberrypi/openocd

My procedure is:
1. git clone the repo from GitHub into my Zephyr workspace
2. Build openocd with a `--prefix=...` configure option to install the new version of openocd locally within my zephyr workspace
3. Use `west build ... -- -DOPENOCD=$PATH_TO_GIT_REPO/build/bin/` to make sure that west and cmake use my newly built openocd binary that has RP2350 support.

Example of cloning and building openocd into a Zephyr workspace directory using a bash shell on Debian 12:

```
$ sudo apt install make libtool pkg-config \
    autoconf automake texinfo libusb-1.0-0-dev libhidapi-dev
$ cd zephyr-workspace
$ git clone https://github.com/raspberrypi/openocd.git
$ cd openocd
$ ./bootstrap
$ ./configure --prefix=$(pwd)/build
$ make
$ make install
```

Then, to make sure cmake finds the right binary, do something like:

```
$ cd zephyr-workspace
$ source .venv/bin/activate
(.venv) $ cd zephyr
(.venv) $ west build -b rpi_pico2/rp2350a/m33  \
    samples/subsys/shell/shell_module --       \
    -DOPENOCD=$(pwd)/../openocd/build/bin/openocd
```

binEncoder, 2025-02-18:
Thanks for your experiment and patient reply.
You pointed out how to use the updated openocd, which is very useful.
However, I still hope that Zephyr can update openocd to facilitate everyone to use the debugger function.
Do you know who else can help?

samblenny, 2025-02-18:
I'm not involved in maintaining the sdk-ng toolchain, so I don't know.
You could try subscribing to the #772 "Upgrade to OpenOCD 0.12.0" issue which was recently created as a parent issue of this one.

russkel, 2025-03-16:
Looks like it will finally be in the next SDK version! 🎊 

Due for an alpha version any day now... 🤞

closed 2025-05-24 by nashif