zephyrproject-rtos/sdk-ng #861: "west sdk install" fails with wget2

aerodinamica, 2025-02-13:
I recently installed Zephyr on Fedora 41 using the "Getting started" guide. I am aware that the guide is primarily intended for Ubuntu users, but with some slight renaming of the required packages, things went smooth. That is, except for the "wst sdk install" command. The script that is used (setup.sh) is expecting wget version 1, and my distro is using wget2, resulting in the following error:

Installing 'aarch64-zephyr-elf' toolchain ...
Unknown option 'show-progress'
ERROR: Toolchain download failed

This is due to using the show-progress option for wget. When changing to "force-progress", the script worked, although I am not sure if that retains compatibility with wget version 1 - perhaps the progress option should be left out entirely.

kartben, 2025-02-13:
See #771

closed 2025-02-13 by kartben

aerodinamica, 2025-02-13:
Oh, sorry! Did a quick search before posting but found nothing. Seems that this will heal by itself. Thanks for the quick reply!

kartben, 2025-02-13:
> Oh, sorry! Did a quick search before posting but found nothing. Seems that this will heal by itself. Thanks for the quick reply!

No worries at all!