zephyrproject-rtos/sdk-ng #881: Installation on Fedora linux 42-beta: buffer overflow detected

scottwcpg, 2025-03-24:
Hi,
I do all my zephyr development on Fedora linux. Since F42-beta is released I wanted to check if they broke zephyr tools.
Looks like they did.
The first parts of zephyr development environment setup installed with no errors.  The sdk install is generating a buffer overflow.  The default python is 3.13.2. I also tried the SDK install in a python3.12 virtual environment and got the same failure.
Ran it this way to get some output: 
$ ./zephyr-sdk-x86_64-hosttools-standalone-0.9.sh -S -D -y -d .
Zephyr Yocto Toolchain SDK installer version 0.9
================================================
+ '[' '' = 1 ']'
+ '[' . = '' ']'
++ echo .
++ sed 's/ /\\ /g'
+ eval target_sdk_dir=.
++ target_sdk_dir=.
+ '[' -d . ']'
++ cd .
++ pwd
+ target_sdk_dir=/home/void/zephyr-sdk-0.17.0
+ '[' 28 -gt 2048 ']'
+ '[' '' = 1 ']'
++ echo /home/void/zephyr-sdk-0.17.0
++ grep ' '
+ '[' -n '' ']'
+ '[' -e /home/void/zephyr-sdk-0.17.0/environment-setup-core2-32-zephyr-linux ']'
+ printf 'You are about to install the SDK to "/home/void/zephyr-sdk-0.17.0". Proceed [Y/n]? '
You are about to install the SDK to "/home/void/zephyr-sdk-0.17.0". Proceed [Y/n]? + default_answer=y
+ '[' Y = '' ']'
+ echo Y
Y
+ '[' Y '!=' Y -a Y '!=' y ']'
+ mkdir -p /home/void/zephyr-sdk-0.17.0
+ '[' '!' -x /home/void/zephyr-sdk-0.17.0 -o '!' -w /home/void/zephyr-sdk-0.17.0 -o '!' -r /home/void/zephyr-sdk-0.17.0 ']'
+ printf 'Extracting SDK...'
Extracting SDK...+ '[' tar.xz = zip ']'
+ tail -n +380 ./zephyr-sdk-x86_64-hosttools-standalone-0.9.sh
+ tar mxJ -C /home/void/zephyr-sdk-0.17.0 --checkpoint=.2500
.................+ echo done
done
+ printf 'Setting it up...'
Setting it up...+ real_env_setup_script=
++ ls /home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
+ for env_setup_script in `ls $target_sdk_dir/environment-setup-*`
+ grep -q OECORE_NATIVE_SYSROOT= /home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
+ real_env_setup_script=/home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
+ sed -e s:/opt/zephyr-sdk/0.9:/home/void/zephyr-sdk-0.17.0:g -i /home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
+ '[' -n /home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux ']'
+ env_setup_script=/home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
+ xargs --version
++ cat /home/void/zephyr-sdk-0.17.0/environment-setup-x86_64-pokysdk-linux
++ grep OECORE_NATIVE_SYSROOT=
++ cut -d= -f2
++ tr -d '"'
+ native_sysroot=/home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux
++ find /home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/lib -name 'ld-linux*'
+ dl_path=/home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2
+ '[' /home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 = '' ']'
++ find /home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux -type f '(' -perm -0100 -o -perm -0010 -o -perm -0001 ')' -printf ''\''%h/%f'\'' '
+ executable_files=''\''/home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-mipsel'\''
...
++ mktemp -d
+ tdir=/tmp/tmp.xog80vVumu
+ '[' x/tmp/tmp.xog80vVumu = x ']'
+ cat
+ mv /tmp/tmp.xog80vVumu/relocate_sdk.sh /home/void/zephyr-sdk-0.17.0/relocate_sdk.sh
+ chmod 755 /home/void/zephyr-sdk-0.17.0/relocate_sdk.sh
+ rm -rf /tmp/tmp.xog80vVumu
+ '[' 1 = 1 ']'
+ /home/void/zephyr-sdk-0.17.0/relocate_sdk.sh
/home/void/zephyr-sdk-0.17.0/relocate_sdk.py:30: SyntaxWarning: invalid escape sequence '\.'
  old_prefix = re.compile(b("/opt/zephyr-sdk/0\.9"))
+ '[' 0 -ne 0 ']'
+ for replace in "$target_sdk_dir -maxdepth 1" "$native_sysroot"
+ find /home/void/zephyr-sdk-0.17.0 -maxdepth 1 -type f
+ grep ':.*\(ASCII\|script\|source\).*text'
+ xargs -n100 file
+ awk '-F: ' '{printf "\"%s\"\n", $1}'
+ grep -Fv -e /home/void/zephyr-sdk-0.17.0/environment-setup- -e /home/void/zephyr-sdk-0.17.0/relocate_sdk -e /home/void/zephyr-sdk-0.17.0/post-relocate-setup -e /home/void/zephyr-sdk-0.17.0/zephyr-sdk-x86_64-hosttools-standalone-0.9.sh
+ xargs -n100 sed -i -e s:/opt/zephyr-sdk/0.9:/home/void/zephyr-sdk-0.17.0:g -e 's:^#! */usr/bin/perl.*:#! /usr/bin/env perl:g' -e 's: /usr/bin/perl: /usr/bin/env perl:g'
+ for replace in "$target_sdk_dir -maxdepth 1" "$native_sysroot"
+ find /home/void/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux -type f
*** buffer overflow detected ***: terminated
xargs: file: terminated by signal 6
sed: no input files
+ '[' 123 -ne 0 ']'
+ echo 'Failed to replace perl. Relocate script failed. Abort!'
Failed to replace perl. Relocate script failed. Abort!
+ exit 1

scottwcpg, 2025-03-29:
Examined zephyr-sdk-x64_64-hosttools-standalone-0.9.sh
Crash occurs when script runs find and then calls file for each found file
Manually ran F42b file on one of the found artifacts, $HOME/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-mipsel.  Something is wrong with the file utility or libmagic from glibc?
F42b has file-5.45 and glibc 2.41
Are there any instruction for building the SDK on a local machine?

schmidma, 2025-06-12:
I am not using this SDK, but stumbled upon this issue, having similar problems in our own yocto based SDK (https://github.com/hulks/meta-nao).

I think the initial problem is caused by a bug in file (reported here: https://bugs.astron.com/view.php?id=638). A fix was discussed and backporting to fedora repos as well: https://src.fedoraproject.org/rpms/file/pull-request/36 https://src.fedoraproject.org/rpms/file/pull-request/37