zephyrproject-rtos/sdk-ng #890: mips: mips-zephyr-elf-gcc Generates Stack Overflow in Parameter Storage with -O0

LuoZhongYao, 2025-04-17:
Component: Target - MIPS

Version: GCC 12.2.0 (Zephyr SDK 0.17.1-rc1)

Target: MIPS32, ELF32 big-endian, O32 ABI

Severity: Critical (Stack corruption in embedded systems)

Keywords: MIPS, MIPS16, stack overflow, codegen, Zephyr, O32 ABI, no optimization

## Summary

The mips-zephyr-elf-gcc compiler (GCC 12.2.0, Zephyr SDK 0.17.1-rc1) generates incorrect MIPS assembly code for a function (test) when compiled with -O0, potentially under MIPS16 mode. The generated code stores parameters (a0 and a1) outside the allocated stack frame, causing stack overflow. 
## Steps to Reproduce
```c
int test1(int a, int b)
{
	return a + b;
}

void test(int a, int b)
{
	int c = test1(a, b);
}
```

```bash
$ ~/.local/zephyr-sdk/mips-zephyr-elf/bin/mips-zephyr-elf-gcc -O0 -c test.c -mips16
$ ~/.local/zephyr-sdk/mips-zephyr-elf/bin/mips-zephyr-elf-objdump -S test.o
```
```asm
test.o:     file format elf32-bigmips

Disassembly of section .text:

00000000 <test1>:
   0:	63ff      	addiu	sp,-8
   2:	d101      	sw	s1,4(sp)
   4:	673d      	move	s1,sp
   6:	d982      	sw	a0,8(s1)
   8:	d9a3      	sw	a1,12(s1)
   a:	9962      	lw	v1,8(s1)
   c:	9943      	lw	v0,12(s1)
   e:	e349      	addu	v0,v1,v0
  10:	65b9      	move	sp,s1
  12:	9101      	lw	s1,4(sp)
  14:	6301      	addiu	sp,8
  16:	e820      	jr	ra
  18:	6500      	nop
  1a:	6500      	nop

0000001c <test>:
  1c:	63fc      	addiu	sp,-32
  1e:	677f      	move	v1,ra
  20:	d307      	sw	v1,28(sp)
  22:	d106      	sw	s1,24(sp)
  24:	0104      	addiu	s1,sp,16
  26:	d984      	sw	a0,16(s1)
  28:	d9a5      	sw	a1,20(s1)
  2a:	9965      	lw	v1,20(s1)
  2c:	9944      	lw	v0,16(s1)
  2e:	67a3      	move	a1,v1
  30:	6782      	move	a0,v0
  32:	1800 0000 	jal	0 <test1>
  36:	6500      	nop
  38:	d940      	sw	v0,0(s1)
  3a:	6500      	nop
  3c:	65b9      	move	sp,s1
  3e:	9703      	lw	a3,12(sp)
  40:	9102      	lw	s1,8(sp)
  42:	6302      	addiu	sp,16
  44:	ef00      	jr	a3
  46:	6500      	nop

```
## Actual Behavior
- The function allocates a 32-byte stack frame (`addiu sp,-32`).
- It sets `s1 = sp + 16` and stores parameters `a0` and `a1` at `s1 + 16` (i.e.,` sp + 32`) and `s1 + 20` (i.e., `sp + 36`).
- These addresses are outside the allocated stack frame (valid range: `sp to sp + 31`), causing stack overflow.

dkalowsk, 2025-04-17:
Clarifying that test function here is not related to the Zephyr ztest suite, but a custom built test file.  Thus it avoids the Kconfig of https://github.com/zephyrproject-rtos/zephyr/blob/2f9faa0a93ccc486db3e165fa1f98a054dc35eb7/subsys/testsuite/ztest/Kconfig#L111

keith-packard, 2025-04-17:
Have you tried later GCC releases?

LuoZhongYao, 2025-04-18:
I downloaded [Codescape.GNU.Tools.Package.2024.10-02.for.MIPS.IMG.Bare.Metal.CentOS-6.x86_64.tar.gz](https://codescape.mips.com/components/toolchain/2024.10-02/downloads.html ) it also has the same problem。 But it is older。It is not easy to find binary compilers for mips

LuoZhongYao, 2025-04-18:
`mips64-linux-gnu-gcc` (GCC) 15.0.1 20250329 (Red Hat Cross 15.0.1-0)`,same problem