zephyrproject-rtos/sdk-ng #892: newlib 3.3.0 is vulnerable to CVE-2021-3420

tfloch, 2025-04-24:
Hi!

newlib version used by all sdk-ng version, is vulnerable to [CVE-2021-3420](https://nvd.nist.gov/vuln/detail/CVE-2021-3420).
CVSSv3 score 9.8.

You should newlib submodule to version 4.0.0, or backport the security patch in 3.3.0 branch.
See PR https://github.com/zephyrproject-rtos/newlib-cygwin/pull/8

kartben, 2025-04-24:
Thanks for the report but please follow our processes for responsible vulnerability reporting, as opening an issue on a public tracker is generally not the best way to proceed, as I'm sure you can imagine :)
https://docs.zephyrproject.org/latest/security/reporting.html

tfloch, 2025-04-24:
Thanks @kartben.

You can delete the issue, I will submit using https://docs.zephyrproject.org/latest/security/reporting.html process in a minute.

kartben, 2025-04-24:
Thank you! Also probably don't want your PR and commit actually mentioning the CVE... but it's a bit too late now of course.

tfloch, 2025-04-24:
@kartben I tried to send the email using 2 differents email adresses, but received a Mail Delivery Failure.

> 550 5.0.350 Remote server returned an error -> 550 Message delivered to [vulnerabilities@zephyrproject.org]
> (mailto:vulnerabilities@zephyrproject.org); Can't send mail - all recipients were rejected: 550 #5.1.0 Address rejected.

tfloch, 2025-04-24:
Submitted with https://github.com/zephyrproject-rtos/zephyr/security form

closed 2025-04-24 by tfloch