zephyrproject-rtos/sdk-ng #896: Ship only one C library (picolibc) and drop newlib from the SDK

nashif, 2025-04-29:
We currently ship newlib in multiple configuration + picolibc. Picolibc is the default and others are provided mostly for historical and legacy reasons. Proposal is top drop newlib and focus on delivering the SDK only with picolibc with the benefit of reducing complexity and reducing size and build time for the SDK.

keith-packard, 2025-04-30:
Sounds like the core-isa.h for that target is promising support for a feature which isn't present in that target?

nashif, 2025-04-30:
ok, looks like the overlays are not being used correctly for xtensa toolchains in picolibc, if I copy the core-isa.h dmanually, it builds fine

nashif, 2025-04-30:
> ok, looks like the overlays are not being used correctly for xtensa toolchains in picolibc, if I copy the core-isa.h dmanually, it builds fine

adding overlay for picolibc fixes the issue.

ok, next one :)

```
/build-workspace/build-aarch64-zephyr-elf/.build/aarch64-zephyr-elf/src/gcc/libgcc/soft-fp/eqtf2.c -fvisibility=hidden -DHIDE_EXPORTS
[ALL  ]    In file included from /home/nashif/zephyrproject/build-workspace/output/aarch64-zephyr-elf/aarch64-zephyr-elf/include/machine/ieeefp.h:12,
[ALL  ]                     from /home/nashif/zephyrproject/build-workspace/output/aarch64-zephyr-elf/aarch64-zephyr-elf/include/sys/config.h:32,
[ALL  ]                     from /home/nashif/zephyrproject/build-workspace/output/aarch64-zephyr-elf/aarch64-zephyr-elf/include/sys/cdefs.h:45,
[ALL  ]                     from /home/nashif/zephyrproject/build-workspace/output/aarch64-zephyr-elf/aarch64-zephyr-elf/include/sys/auxv.h:39,
[ALL  ]                     from /home/nashif/zephyrproject/build-workspace/build-aarch64-zephyr-elf/.build/aarch64-zephyr-elf/src/gcc/libgcc/config/aarch64/cpuinfo.c:28:
[ERROR]    /home/nashif/zephyrproject/build-workspace/output/aarch64-zephyr-elf/aarch64-zephyr-elf/include/sys/features.h:28:10: fatal error: picolibc.h: No such file or directory
[ALL  ]       28 | #include <picolibc.h>
[ALL  ]          |          ^~~~~~~~~~~~
[ALL  ]    compilation terminated.

```

with `aarch64-zephyr-elf`

keith-packard, 2025-05-01:
I'll have to wait until Sunday to take a serious look at this -- I'm traveling and my build machine has taken itself offline without any remote console configured. It looks like there's some issue with the crosstool-ng configuration that uses picolibc as the primary C library. I haven't tried the Zephyr version of those patches in a while...

adigie, 2025-05-01:
picolibc libstd++ is built with exceptions enabled. Removing newlib (nano) would increase size of the C++ applications by ~4kB (#860). I think this should be considered before making such change.

keith-packard, 2025-05-01:
> picolibc libstd++ is built with exceptions enabled. Removing newlib (nano) would increase size of the C++ applications by ~4kB ([#860](https://github.com/zephyrproject-rtos/sdk-ng/issues/860)). I think this should be considered before making such change.

We could consider changing the picolibc configuration if we're willing to merge the libstdc++ module for Zephyr. That seems blocked at this point. https://github.com/zephyrproject-rtos/zephyr/pull/80762

Damian-Nordic, 2025-05-08:
> We could consider changing the picolibc configuration if we're willing to merge the libstdc++ module for Zephyr. That seems blocked at this point. [zephyrproject-rtos/zephyr#80762](https://github.com/zephyrproject-rtos/zephyr/pull/80762)

I understand that we didn't want to include libstdc++ binaries with and without exceptions enabled not to increase the toolchain size, but if newlib is to be removed and so is libstdc++ built for newlib, then maybe two variants of libstdc++ for picolibc wouldn't be a big problem? At least until the support for libstdc++ as a module is merged.

keith-packard, 2025-05-08:
> I understand that we didn't want to include libstdc++ binaries with and without exceptions enabled not to increase the toolchain size, but if newlib is to be removed and so is libstdc++ built for [...]

That's a reasonable idea; we effectively do this today with newlib/newlib-nano -- newlib gets exceptions, newlib-nano does not.

I'm not sure how best to approach this though; we could try adding another multilib selector and see if that 'just works'. Alternatively, installing the two versions in separate directories or using different library names, although that would require some crosstool-ng hacking.

carlescufi, 2025-05-12:
@frkv @tomi-font will TF-M build fine without newlib?

tomi-font, 2025-05-12:
@Vge0rge any idea?

frkv, 2025-05-12:
I find no direct link to newlibc in the build-system

TF-M provides memcmp, memcpy, memset, printf, vprintf via a runtime library level.

Additionally it provides heap support via Mbed TLS (which can defer down to hand-implemented buffers)

carlescufi, 2025-05-12:
> I find no direct link to newlibc in the core TF-M build-system, beyond some usages of spec-strings in `toolchain_GNUARM.cmake` (`-specs=nano.specs`, `-specs=nosys.specs`) as well as a library dependency for `-lc -lnosys` in `psa-arch-tests`
> 
> TF-M provides `memcmp`, `memcpy`, `memset`, `memmove`, `printf`, `vprintf` via a runtime library level. Additionally it provides heap support via Mbed TLS (which can defer down to hand-implemented buffers)

Understood. So, to me this sounds like a "no, TF-M does not use newlib".

tokangas, 2025-05-14:
While updating an application from newlib to picolibc, I ran into a linking problem. It seems to be caused by calling `mallinfo()`. After I had figured that out, I found this bug: https://github.com/zephyrproject-rtos/zephyr/issues/59424

`mallinfo()` should also work with picolibc, right?

keith-packard, 2025-05-14:
mallinfo isn't provided in the Zephyr malloc implementation which is used when you use picolibc. If you reference it, you'll end up pulling in the picolibc version which isn't useful.

tokangas, 2025-05-15:
> mallinfo isn't provided in the Zephyr malloc implementation which is used when you use picolibc. If you reference it, you'll end up pulling in the picolibc version which isn't useful.

Thanks, now it makes sense. It would be nice, though, that calling `mallinfo()` would cause an error which would be more informative. Currently, it's quite hard to figure out the root cause. And I guess it should be possible to add `mallinfo()` support, if desired.

keith-packard, 2025-05-15:
Yeah, it might be nice to not include malloc and friends in the SDK.

tokangas, 2025-05-16:
One of our applications has used `mallinfo()` to get statistics of `malloc()` heap usage. After migrating to picolibc, `mallinfo()` is not available. So, I tried to use `sys_heap_runtime_stats_get()` to get statistics from `z_malloc_heap` instead, but it turns out that it does not work either, because `z_malloc_heap` is `static`.

@keith-packard Is there some method to get `malloc()` memory usage statistics with picolibc? I think there should be.

closed 2025-05-26 by nashif

stephanosio, 2025-07-25:
For tracking purposes, newlib was removed in the commit https://github.com/zephyrproject-rtos/sdk-ng/commit/9ecd76a0e5c617f3d4f09ca42571161ec18fea70.