zephyrproject-rtos/sdk-ng #967: QEMU x86 and x86_64 reset terminal with SDK 0.17

dcpleung, 2025-07-21:
QEMU x86 and x86_64 on latest SDK v0.17 now reset the terminal when they run, where the QEMU in previous SDKs did not. This is a bit annoying because you can no longer scroll back to see the build process when you run `west build -t run`.

This is due to the SeaBIOS (`bios.bin` and `bios-256k.bin`) being built with QEMU using the upstream commit, instead of our own fork.

stephanosio, 2025-07-21:
Ah, yes. https://github.com/zephyrproject-rtos/sdk-ng/commit/1bf6f4affb772d50c19af6c1a5a8e555eb4a2be3 got missed during the latest QEMU update. I will look into this.

closed 2025-07-22 by stephanosio

stephanosio, 2025-07-22:
Keeping this issue open until https://github.com/zephyrproject-rtos/sdk-ng/pull/971 is backported to `v0.17-branch.`